ebastler/marbastlib #1: Choc .kicad_sym file not present in Untested branch?

fatihG, 2022-02-18:
I tried to add the library files to KiCad after following the instructions on how to add the submodule. 

After I did all the steps I was unable to locate the symbol library for the choc components. 
The MX and Various .kicad_sym files are present and I am able to add those to KiCad. 

Since I am planning on using chocs for my board, I figured using that library would be the easier option.

ebastler, 2022-02-19:
The only symbol I have in the mx.kicad_sym is a symbol for the stabilizers - I usually just use generic kicad pushbotton symbols for key switches, that's why the chocs don't have a dedicated symbol lib.

fatihG, 2022-02-20:
Makes sense. 

I am very noob to all this PCB stuff, I forgot you had to manually assign footprints to symbols.

ebastler, 2022-03-04:
This has been addressed, together with some other issues, in today's patch. Thanks for pointing it out!
If you want to update to the latest lib:
`git submodule update --remote`, but watch out as it introduced some other changes as well (Mainly all hotswap and  reverse mount LED footprints are now flipped, and must manually be flipped back).

closed 2022-03-04 by ebastler