ebastler/marbastlib #17: Question about Choc stabilizer footprint

ubikitty, 2023-08-01:
The choc stabilizer footprint cuts two holes in the pcb.
I would like to know what they are used for and if they are mandatory?

[image]

ebastler, 2023-08-09:
Hi! The choc stabs themselves protrude through the PCB, so sadly they are mandatory. pretty annoying...

The big one is for the slider, can't remember what the small one is for. Probably the part holding the wire. 

Ah, one caveat: The slider is offcenter for some weird reason, that makes Choc stabilized keys directional. Can only put them on "the right way", if you rotate the caps 180° they won't fit anymore. If you wanna use any sculpted profile caps (I do not know of any that exist), the stab orientation on the PCB/plate will determine which way the cap fits...

ubikitty, 2023-08-09:
Thank you very much for your answer and your insights on the quirks of the choc stabilizers! Did not notice that the stabilized keys were asymmetrical...

closed 2023-08-09 by ubikitty

ebastler, 2023-08-09:
Found 2 pics of a MBK 2u cap, maybe they are interesting for this topic.

[image]