ebastler/marbastlib #19: SW_MX_HS_1u and mayber other hot swaps have an issue with kicad free rooting plugin

EcceGratum, 2023-08-19:
Hello,

I use the free rooting plugin in kicad and noticed that it tends to create tracks linked to the 2 holes of the hotswap (SW_MX_HS_1u, maybe others) which would not work.

I edited the footprint by changing the "Pad Type" of the 2 holes, i use "NPTH, Mechanical" instead of "Through-hole".
To get rid of the Clearance violation between the pads and the holes, i remove the small portion between the pads and the holes :

[image]

ebastler, 2023-08-20:
Hi! I think you're the first person to use the lib with autorouters, so we mainly focused around that usecase. The plated holes are used for stability, but also as a design element and for easier routing (as they can be used for traces which saves quite some manually placed vias).

I'm sorry it seems to affect your usecase, but the footprints were made the way they are for a reason, and changing them to NPTH would negatively affect other users (incl. us maintainers), so we will keep it the way it is.

Is it really a problem if the Autorouter connects to the PTHs? Since they are electrically connected to the pads, I am using them as contacts on a lot of boards as well.

EcceGratum, 2023-08-21:
Hello,
"Is it really a problem if the Autorouter connects to the PTHs? Since they are electrically connected to the pads, I am using them as contacts on a lot of boards as well."
Oh ok, may explain why the autorouter uses it, i though it would not work and looked a bit weird compared to other boards but if it works that's fine. This is the first board i am making so my knowledge is very limited.

I have no issue changing the footprints locally for my boards as it's quite easy and quick to do. I reported it thinking it could be an issue for other people.

If that's alright, i will close.

closed 2023-08-23 by EcceGratum