ebastler/marbastlib #21: Too small PCB holes for Choc v1

bolleywall, 2023-09-08:
In marbastlib Choc v1 footprints, the diameter of the holes for the smaller mounting pins is 1.7 mm., while Kailh recommends 1.9 mm.

ebastler, 2023-09-08:
Hi! Thank you for pointing this out.

If I remember correctly we opted for this to allow plateless hotswap to work decently without wobbling, and then mistakingly used the footprint for both HS and solder. 

It is tested and works (although the switches will afterwards probably sit loosely in 1.9mm hole PCBs if you remove them from ours). I am considering to keep it for hotswap, but switch to the recommended hole sizes for solder footprints as those do not need the mechanical stability of tight legs. 

On the other hand I feel reluctant to change something tested and working. I will order a testing PCB with my next PCB order (with footprints with 1.7mm, 1.8mm and 1.9mm) and decide.

bolleywall, 2023-09-09:
Thank you for explaining your reasoning!  Now I too need to order a test PCB and compare various hole sizes for both the side posts and the middle one :-)

Apparently some feel 1.9 mm is too large (link for reference: https://old.reddit.com/r/ErgoMechKeyboards/comments/mdvy5w/what_size_through_holes_do_you_use_for_kailh_choc)