ebastler/marbastlib #22: Seeeduino Xiao boards support?

idesignstuff, 2023-09-09:
The Xiao (especially the NRF52840 "BLE" version) boards are an increasingly popular choice for split wireless ZMK keebs. Would you be willing to add them to your footprints/symbols library?

ebastler, 2023-09-09:
Hi!

I was planning to extend the promicorish lib and add (among others) xiao, but did not find the time for it so far - I mostly focused on footprints I needed myself recently as I was short on time.

Can't give any proper ETA, but it is planned :)

idesignstuff, 2023-09-09:
Thanks for considering it!

ebastler, 2023-09-11:
I'm currently beginning to work on them - if you have any recommendations on how to handle the bottom pads... This is a Xiao BLE, will also make a xiao rp2040 and maybe esp32 later on. 

I know that on BLE you need the bottom pads for both flashing and battery. I will make a SMD only version (-> solder paste covered SMD pads, for reflow), but not quite sure how to handle them for THT and combo footprints...

[image]

closed 2023-09-11 by ebastler