ebastler/marbastlib #23: Wrong Pin numbers for MX_SK6812MINI-E Symbol

Bob837217, 2023-09-09:
According to the datasheet copy hosted at: https://datasheet.lcsc.com/lcsc/2305101623_OPSCO-Optoelectronics-SK6812MINI-E_C5149201.pdf
This was last edited on 2021-04-07 with revision "A/O"

Here is the pinout from that PDF:
[image]

Here is the Symbol pinout as seen in KiCAD 7.0.6:
[image]

```
PDF Pinout Numbers:
VDD:    1
DOUT:   2
GND:    3
DIN:    4

Symbol Pinout Numbers:
VDD:    1
DOUT:   2
DIN:    3
GND:    4
```

Symbol uses "VSS", but to my knowledge that means "GND".

ebastler, 2023-09-09:
Hi, this has been done for a reason - the footprint is shared between 2 LEDs with different pin numberings, so the symbol does not follow datasheet numbering, but footprint numbering.

If you compare the 6028 footprint, you will notice it deviates from the datasheet, but in the end the different numbering on symbol and footprint results in a correct pinout - these have been tested.

Bob837217, 2023-09-09:
Ok, thanks for the explanation.  I will mark this as closed.

closed 2023-09-09 by Bob837217