ebastler/marbastlib #28: Gateron KS-27/33 support?

ZoeArch, 2024-02-14:
Gateron is offering low profile switches and their accompanying hotswap sockets, would it be possible to add these to this project?

https://www.gateron.co/products/gateron-low-profile-mechanical-switch-set

ebastler, 2024-02-14:
Hello Zoe,

I would love to, but I do not have access to datasheets, 3D models or similar for those. The switches themselves at least have tech drawings on the product site, but I have no idea about the sockets. I can try messaging Gateron, but I somewhat doubt they will get back to me...

I'll give the switches a shot, at least. Sockets and 3D models when/if I hear back from Gateron.

CosmicCuttlefish, 2024-02-19:
Hey ebastler,

I was looking at gateron's website and found these 3D models: https://www.gateron.com/pages/products
And product specifications: https://www.gateron.com/pages/product-specification
Not sure if this helps but I thought I'd bring it up

ebastler, 2024-02-19:
Huh, how did I miss that! Thanks. Gateron themselves sent me the info via email by now, but that's great to know if I will ever need it again!

CosmicCuttlefish, 2024-02-19:
I'm glad they sent over the info! Do you have an ETA for these footprints? No pressure, just curiosity.

ebastler, 2024-02-19:
Hmm, depends on my spare time, motivation and if I want to push switch/stab 3D models along with it. I think I will do footprints alone for now, and add switch/stab models in a later, larger lib update.

ebastler, 2024-02-20:
Okay, I was motivated... :D

The files are up, but I will not create a new release until I have time to immediately debug and hotfix. This commit changes a ton of things to account for the fact that now there is 2 different MX-style HS sockets, so I remade the entire HS socket file/footprint names etc.

closed 2024-02-20 by ebastler

ebastler, 2024-02-20:
[image]

IIIBlueberry, 2024-07-01:
Need help KitCad 8.0.3 cannot load KS-2P02B01-01 footprint?

`Processing symbol 'MX1:PCM_marbastlib-mx:SW_MX_HS_CPG151101S11_1u'.
Processing symbol 'MX2:PCM_marbastlib-mx:SW_MX_HS_KS-2P02B01-01_1u'.
Add MX1 (footprint 'PCM_marbastlib-mx:SW_MX_HS_CPG151101S11_1u').
Error: Cannot add MX2 (footprint 'PCM_marbastlib-mx:SW_MX_HS_KS-2P02B01-01_1u' not found).
Total warnings: 0, errors: 1.`

Finnitio, 2025-01-02:
Is there a reason for making the hole for the plastic cylinder of the switch 5,25 mm?

[image]

The max, size in the drawing is 5,05 mm. I could not find any explicit size for the pcb hole.