ebastler/marbastlib #30: Could use an update to point to 3D Model files in KiCAD 8.

machineabuse, 2024-02-25:
As above, despite being able to install the libraries within KiCAD 8's PCM, the 3D model locations are still directed to KiCAD 7, requiring manual linking.

One other peculiarity is that within KiCAD 8, the MX Hotswap footprint and model names have an added code string, not sure if this affects only the MX Hotswap footprints or other files as well.

ebastler, 2024-02-29:
Thanks for reporting this!

Will tackle that once I find the time to migrate to kicad 8 myself. I have some larger projects on kc7 and no time to deal with migration. As long as I am on kicad 7 on my work machine, I can not yet switch the lib over. Hope I will get to it this weekend.

machineabuse, 2024-03-01:
You're an absolute sir! ❤

closed 2024-03-01 by machineabuse