ebastler/marbastlib #31: DRC fail on SK6812MINI-E footprint ("Error: Board edge clearance violation (boa...clearance 0.3500 mm; actual 0.2531")

todbot, 2024-03-19:
I'm trying to place the SK6812MINI-E footprints and DRC is giving me a board clearance error for the pads. (see screenshot) 
Is there a way to fix this without reducing the global board clearance or is that what everyone does (I'm fairly new to Kicad, coming from Eagle)

[image]

ebastler, 2024-03-30:
Hi! That error is no problem - you can either get rid of it by lowering edgecut to pad clearance in your board setup,or just ignore it.

I will try to adjust the pads a little when I find time, so it will no longer throw errors in the first place!

todbot, 2024-03-30:
Hi, thanks for responding. I've since found DRU settings for the fabs I'm using and these errors do go away. (Looks like KiCad's defaults are pretty conservative).  Thanks for your patience in me being a newb with KiCad. It's very nice and your library with these complex parts has made the transition much easier. Closing this issue.

closed 2024-03-30 by todbot