ebastler/marbastlib #35: LED_MX_6028R-ROT footprint seem not fit with switch

Halogene116, 2024-07-11:
Dear Mr. marbastlib 
LED_MX_6028R-ROT in MX footprint seem bigger than switch (1u), so I have to use Led 6028 form Choc library.
Beside that, MX stab library seem not fit to switch too. I just use the Kicad for fisrt time so sr if it not true.
And I need a bigger Mec Keyboard than 104 Fullsize, so i need a PCF8575 I/O Expander module for more pins to control many switch and LED. I have searched for symbol and footprint but no hope to get right. Would you mind to make a symbol and footprint for PCF8575 module as picture?

Thank you so much for your library. It helps me so much for make my own keyboard.
[image]

ebastler, 2024-07-11:
Hello! 

I try to address all points - hope I am not missing any.

1) Both footprints are `19.05` by `19.05` mm on the user.drawings layer, as they should be (usual spacing between MX switches is `19.05` by `19.05`):
[image]

I am not quite sure what you mean with the LEDs being larger.

2) The same applies to stabilizers - `19.05` by `x * 19-05` as well, with x being the length of the stabilizers in "units". It all lines up as it is supposed to. Don't let my colors confuse you, I am not using the default KiCad color scheme, so things may look a little different.
[image]

3) Which microcontroller are you using? In general I would rather recommend using a larger controller (like STM32F072RBT6 which I used on E80-1800), those should be able to work with plenty of keys to go way beyond 104 key fullsize. IO expanders lead to increased input delay and hugely increased firmware complexity.

If you still want to use an IO expander, I would rather put the bare chip directly on the PCB. This expander breakout board has barely any components on it, and I honestly have no idea what the voltage regulator and the two transistors are even doing - there is no trace of those in the chip datasheet. Maybe some level shifting circuit for which I see no reason to exist. I have used PCA9555 IO expanders with QMK keyboards before, put the chip directly on the PCB.

If you have detailed measurements for this expander breakout PCB (pin spacing) I can still make a footprint and push it sometime in the next couple of days.

Good luck and enjoy the process of designing your first keyboard! Don't be overwhelmed by the amount of new information early on, it gets easier with time!

Kind regards,
Moritz

Halogene116, 2024-07-12:
Dear Mr. Marbastlib,
Thank you for your info about input delay things. And I need some more advice about my Mec Keyboard.
1. My Keyboard has 30 cols and 6 rows, so total it needs 36 pins for keyboard matrix.
My old plan is use 1 Arduino Micro Pro (Atmega32U4 chip) and some IO expansion module through I2C protocol to control the matrix. But as you said it will delay the input things. So I need an advice to improve it like:
a. Use a module control with over 36 pins. (quite expensive to me, )))
b. Use 2 Arduino Micro Pro modules (or more) as master and slaves model to have enough pins (cheaper than "a" solution) but I don't know if it has input delay things like my old plan.

2. My Keyboard also has RBG leds. I also want to control every single Led as Switch, so I plan to get more Pins to Led matrix as Switch Matrix. So how can I design my control module to get it? And the Leds is not important as Switch so I can accept it delay a little bit than Switchs.

I look forward to hearing more advice from you.
Thank you so much.

ebastler, 2024-07-12:
1) For this many switches I would use a stm32f072rbt chip directly, instead of an Arduino/module. jlcpcb can solder those for cheap for prototypes, and you only need ~5 capacitors and 2 resistors together with the chip, I think. Can not remember exactly. A bit more difficult than using an Arduino, but not a huge problem.
2) Use either SK6812MINI-E (WS2812 protocol, but reverse mounted for keyboard per key RGB) or use one (or more) dedicated LED controllers like ISSI IS31FL3733. The SK6812MINI-E are way easier to wire. The ISSI is more cost efficient for larger PCB production batches. But for small batch sizes and as a beginner, I would recommend SK6812. Marbastlib has the `MX_SK6812MINI-E` symbol with footprint already assigned. Tested and working. So you only need 1 pin on the MCU for all your LEDs.

Halogene116, 2024-07-15:
Dear Mr. Marbastlib,
Thank you for your info. I will apply it to my Keyboard.
And if i don't have stm32f072rbt chip, can i use stm32f104xxx or higher different stm chip.
Thank you so much.

ebastler, 2024-07-22:
You can in theory use any chip from this list (assuming you want to run QMK firmware on your board): https://docs.qmk.fm/compatible_microcontrollers

I just recommended the F072 series since they are the easiest to use - no need for dedicated programming tools, no crystals. Just the microcontroller and a handful of capacitors and resistors.

Something like this works well for smaller amounts of LEDs. Alternate `PCM_marbastlib-mx:LED_MX_6028R` and `PCM_marbastlib-mx:LED_MX_6028R-ROT` between rows so the datalines can "snake" left and right alternatingly.

[image]

If you have more LEDs (and you do), take care of using sufficiently thick power traces, and lay them down in a better way. The Way I did it here with the power traces snaking left and right too would lead to way too long tracks, which need to be very thick.

Something like this might be better suited for your application - pick one spot on the board where you can comfortably put down a thick vertical connection between the power pins of all LED rows.

[image]

Halogene116, 2024-07-26:
Thank you so much for everythings. It really helpful for newbee like me.

closed 2024-07-26 by Halogene116