ebastler/marbastlib #38: KS27 KS33 Hotswap footprint hole too big.

Finnitio, 2025-01-27:
I just received pcbs and the hole diameter is too big with 5.25 mm. The switches are completely lose and only stay loose in place due to the hotswaps.

[image]

ebastler, 2025-02-16:
Thanks for reporting this! Only got to check it out now...

The footprints are following Gaterons recommended landing pattern, which is quite confusing. I wonder how wide I should make the hole. 

[image]

Kailh Choc have a 3.2mm center pole and the recommended footprint states 3.4mm, so 0.2mm "clearance" just like Gateron did here. That seems to work fine on Kailh. I wonder why it does not on Gateron.

Did you measure the actual hole with calipers to make sure the factory did not make a mistake and used the wrong drill bit?

Finnitio, 2025-02-18:
I did not measure them, but considering the other drawing it is obvios that there is some play in the contact pair.